rahul-thakoor/rust_gpiozero #1: ETA on support for Thermometer sensors?

BlackHoleFox, 2018-07-12:
I found your book that you wrote up on physical computing. (Well done btw) Floating over to this repo for the crate I saw that other common sensor support is on a todo. Is there an ETA for when some would be added? Something like this would be perfect for a thermostat control application that im building.

Thanks!

rahul-thakoor, 2018-07-14:
hey there @BlackHoleFox ! thanks a lot. Unfortunately, at this point, I'm unable to give an ETA. I am considering some under the hood changes regarding some dependencies used. Is there any specific temperature sensor you would like to have support for?

BlackHoleFox, 2018-07-14:
I was looking at one of these https://www.adafruit.com/product/374 for my project to hook up to my raspberry pi. Regarding the ETA, no rush. Plenty of other parts to work on before I get to sensor type stuff.