rahul-thakoor/rust_gpiozero #11: Looking for contributors and maintainers

rahul-thakoor, 2020-07-09:
I could use some help maintaining or adding features to rust_gpiozero. It is intended to be a port of the python-based [gpiozero](https://gpiozero.readthedocs.io/en/stable/) library. I would be grateful for any kind of help. Thanks

AldaronLau, 2020-07-15:
I was actually thinking of starting something like this, as I have some python code I wanted to re-write in Rust - but I haven't got around to it yet.  I suppose instead, I could help maintain this project.  Is there a reason it's licensed GPL?  This might limit the number of people who use the library, rather than something like the ZLib license which is made specifically for statically-linked libraries.

karisair, 2020-07-15:
I’m learning and could also pitch in!

azjkjensen, 2020-07-20:
I'd be happy to contribute here as well!

rahul-thakoor, 2020-07-20:
hey @AldaronLau @karisair @azjkjensen thanks a lot!

Any suggestions on how to proceed?

rahul-thakoor, 2020-07-20:
@AldaronLau no particular reason for choosing that license. perhaps might make more sense to got with MIT/Apache-2.0 as other rustlang crates? thoughts?

AldaronLau, 2020-07-20:
I think I'll have more ideas on how to proceed with the direction of the crate after I start using it in a personal project.  I will review all of the code and see if anything stands out.  I think for the current open issues and the one current pull request, we could assign one of us to work on each depending on who's interested in what.  As for adding additional maintainers (which I'm defining as people who merge pull requests into master, respond to issues, and can release new versions on crates.io), you'll probably want to make sure that they're people who have similar goals with you for this project before giving collaborator status.  That way you won't always have to be involved with every change made to this library (which I'm guessing is your goal).

As for the license, I think MIT/Apache-2.0 is fine and consistent with the rest of the ecosystem, but I personally choose Zlib/Apache-2.0 for crates I've started - as it's essentially the same with the minor exception that Zlib/Apache-2.0 doesn't require the license to be included in binary applications that depend on the library.  But, either one is good.

azjkjensen, 2020-07-21:
I'm fine with dividing up the existing work. Let me know which issue/PR to take.

rahul-thakoor, 2020-07-22:
thanks @azjkjensen @AldaronLau 

I will add issues for further work to be done. feel free to add issues you see relevant

rahul-thakoor, 2020-07-22:
Feel free to comment on those you would be willing to work on

@azjkjensen perhaps you can help test the current open PR?

azjkjensen, 2020-07-23:
Sounds great, I'll check it out.

pierwill, 2021-01-24:
I'm going to look into implementing `bounce_time` for button presses...

dcroote, 2021-12-19:
I'd be happy to help move from Travis CI to GitHub Actions and get the builds working

rahul-thakoor, 2021-12-20:
thanks @dcroote that would be great. we should also automatically publish to crates.io

dcroote, 2022-01-09:
@rahul-thakoor - if you're willing, I'd be happy to assist in basic maintenance and some PR reviews when I have the availability. I recently used this package to build a small project on my pi zero w and would like to support others wishing to contribute and do the same.

rahul-thakoor, 2022-01-10:
hey @dcroote awesome! i would really appreciate that :)