rahul-thakoor/rust_gpiozero #21: Questions regarding this project!

Joe23232, 2021-02-01:
Hi, I have some questions regarding this project.

1. Is this project abandoned as it hasn't been updated for a long time?
2. Would this crate also work on the [Odroid](https://www.hardkernel.com/) and similar mini computers with GPIO ports?
3. Is this based on [`embedded-hal`](https://github.com/rust-embedded/embedded-hal) crate?
4. Is this library stable? If not, would the API have significant changes overtime?

rahul-thakoor, 2021-02-01:
hey @Joe23232 
Thanks for the questions :)

1. I could use some help maintaining the project for sure. see #11 
2. This crate is very raspberry-pi specific. 
3. This crate uses https://github.com/golemparts/rppal under the hood but rust_gpiozero hasn't been updated to use the embedded-hal feature it now supports

> I would suggest using [linux embedded hal](https://docs.rs/linux-embedded-hal/0.3.0/linux_embedded_hal/) for a generic approach and something that would work on the Odroid
4. This library is not stable yet.
5. Not all parts of gpiozero lib has been ported. PRs are welcomed 😄

Joe23232, 2021-02-01:
@rahul-thakoor 

> Thanks for the questions :)

:)

1. I wish I could help you out mate, I wish I could be more technical I am still a noob :)
2. I thought that the Odroid also has GPIO pins so why wouldn't this crate work just out of curiosity?
3. I see mate :)

> I would suggest using linux embedded hal for a generic approach and something that would work on the Odroid

Is this crate more difficult to use over `raspberry-pi`?

4. I see mate
5. Ah right I see mate :)

closed 2021-05-06 by rahul-thakoor