rahul-thakoor/rust_gpiozero #22: Panic on simple blink example

bammi, 2021-02-19:
Raspberry PI4 4GB, Raspberry PI OS up to date version
Simple blink program results in Panic (back trace below). I am trying to learn Rust, cant figure out what the issue is. Simple hello_world type programs run fine, so looks like Rust install is fine. Tried with sudo no difference. Appreciate advice

Attached (had to add .txt extensions otherwise Github does not allow me to attach)
Backtrace
Cargo.toml
main.rs

[backtrace.txt](https://github.com/rahul-thakoor/rust_gpiozero/files/6011796/backtrace.txt)

[Cargo.toml.txt](https://github.com/rahul-thakoor/rust_gpiozero/files/6011803/Cargo.toml.txt)

[main.rs.txt](https://github.com/rahul-thakoor/rust_gpiozero/files/6011811/main.rs.txt)

bammi, 2021-02-19:
In Cargo.toml if i change the dependency to version 0.1.0  it works and runs as expected. So it has something to do with 0.2.0
(i had to change led to mut with 0.1.0)

[dependencies]
#rust_gpiozero = "0.2.0"
rust_gpiozero = "0.1.0"

Any help appreciated.

cheers,
bammi

rahul-thakoor, 2021-02-20:
Hi, it seems that the current version of https://github.com/golemparts/rppal used doesn't have support for pi4.

I have tried to update it to 0.11.3 here https://github.com/rahul-thakoor/rust_gpiozero/tree/rppal-0.11.3 to see if just changing the dependency fixes it. 

Can you try to change your `Cargo.toml` to:

```toml
[dependencies]
#rust_gpiozero = "0.2.0"
rust_gpiozero = { git = "https://github.com/rahul-thakoor/rust_gpiozero", branch = "rppal-0.11.3" }
```

let me know if it works. thanks

bammi, 2021-02-20:
Thanks Rahul,
Unfortunately it did not fix the problem, I get the same exception as I did with 0.2.0. Traceback below. My toml file attached 
[Cargo.toml.txt](https://github.com/rahul-thakoor/rust_gpiozero/files/6015530/Cargo.toml.txt)
 for your reference

Any further advise much appreciated. Maybe instruction on how I can build rppal locally on my RPi4 and use it -- I am learning Rust as we speak, still unsure of how to do detailed stuff like this, but learning. Have substantial CS background.

cheers,
bammi

The traceback
pi@pc-raspi41:~/src/rust/blink $ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `target/debug/blink`
thread 'main' panicked at 'Io(Os { code: 2, kind: NotFound, message: "No such file or directory" })', /home/pi/.cargo/git/checkouts/rust_gpiozero-f2747305eab56c2e/2f4b39f/src/output_devices.rs:99:23
stack backtrace:
   0: rust_begin_unwind
             at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/panicking.rs:493:5
   1: std::panicking::begin_panic_fmt
             at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b/library/std/src/panicking.rs:435:5
   2: rust_gpiozero::output_devices::OutputDevice::new
             at /home/pi/.cargo/git/checkouts/rust_gpiozero-f2747305eab56c2e/2f4b39f/src/output_devices.rs:99:23
   3: rust_gpiozero::output_devices::LED::new
             at /home/pi/.cargo/git/checkouts/rust_gpiozero-f2747305eab56c2e/2f4b39f/src/output_devices.rs:281:41
   4: blink::main
             at ./src/main.rs:8:15
   5: core::ops::function::FnOnce::call_once
             at /home/pi/.rustup/toolchains/stable-armv7-unknown-linux-gnueabihf/lib/rustlib/src/rust/library/core/src/ops/function.rs:227:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

rahul-thakoor, 2021-02-21:
I have just tried running the blink example on a Pi4. I am able to toggle an LED attached to GPIO17 or GPIO18.

Can you try with GPIO17? `GPIO18` has several alt modes and perhaps it is being used for other purposes on your system?

Are you running this on RaspberryPi OS?

bammi, 2021-02-21:
Thanks Rahul
Unfortunately, that did not work. I tried all kinds of GPIO pins, 17/18/12 ....  All of them throw the same exception as in my previous message. I am running the latest Raspberry PI OS version 20.10 with all updates, this is a RPI 4/4GB, uname -a reports:
     Linux pc-raspi41 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

BTW: I tried RPPAL directly,  a Blink written to that API without gpiozero involved, and it too panics/exceptions. I opened an issue on the RPPAL github. Again I tried that on several GPIO pins.

Not quite sure where to go with this, but I will try a fresh RPI 4/8GB that is arriving tomorrow in case there is something weird going on with this RPI.

Also, everything works just fine with gpiozero 0.1.0 so that leads me to believe that I don't have a hardware issue.

thanks again, any other thoughts much appreciated,
cheers,
bammi

bammi, 2021-02-25:
Rahul,
In order to get the rppal blink example to work I had to use link to the git directly : See here
https://github.com/golemparts/rppal/issues/73#issuecomment-780329963

by specifying 
rppal = { git = "https://github.com/golemparts/rppal.git" }
in the dependency section of Cargo.toml

How  does that translate into doing the same for rust_gpiozero (dont know rust enough to figure it out). Any help appreciated,

cheers,
bammi

rahul-thakoor, 2021-02-27:
hello @bammi , I have created a branch that uses `rppal` source directly

Can you try to change your Cargo.toml to:
```toml
[dependencies]
rust_gpiozero = { git = "https://github.com/rahul-thakoor/rust_gpiozero", branch = "rppal-edge" }
```

**Note**: It seems that `rppal` has introduced breaking changes. I have not tested this yet. Would appreciate your feedback regarding using latest `rppal` changes.

Thanks

bammi, 2021-02-27:
Thanks You Rahul,
That worked a treat!!

will now move on beyond blink!
Really appreciate you help,

cheers,
bammi

closed 2021-02-27 by bammi