rahul-thakoor/rust_gpiozero #27: Unmable to make LED blink.

parkeradam, 2021-07-20:
I have taken the code from https://rahul-thakoor.github.io/physical-computing-rust/step_6.html#flashing-an-led
And tried to use the version not with a build method. The LED turns on, and does not turn off. 

I added printlines to view as shown here 
``` 
extern crate rust_gpiozero;

use rust_gpiozero::*;
use std::{thread::sleep, time::Duration};

fn main() {
    let mut led = LED::new(17);

    loop {
        println!("Turning on");
        led.on();
        println!("Is on");
        sleep(Duration::from_secs(1));
        println!("Turning off");
        led.off();
        println!("is on");
    }
}

```

Still no effect. If i kill the program and simply run the LED off, it turns off.

Using a Raspberry Pi Zero W with GPIO pins added on [found here](https://thepihut.com/products/gpio-hammer-header-solderless?variant=28950933073).

Crate version is 0.2.1

Any further information please feel free to ask. 

Thanks in advance for any assistance/investigation.

dcroote, 2021-12-30:
Hi @parkeradam - I've tested your code and found the issue: you need to add a `sleep` step after the LED is turned off. What is currently happening is that the loop ends with `led.off()`, then immediately restarts with `led.on()`, which results in the LED never turning off visually since this happens so quickly.

dcroote, 2022-01-29:
I'm closing this issue given it has been addressed with the above comment. Feel free to reopen if that is not the case.