rahul-thakoor/rust_gpiozero #30: Hi, is this project dead?

Joe23232, 2021-10-31:
Hi, is this project dead?

rahul-thakoor, 2021-11-09:
Looking for maintainers and contributors, see #11

Joe23232, 2021-11-10:
@rahul-thakoor I wish I can be a developer myself. Oh well.

dcroote, 2022-01-29:
The project is not dead :-)

closed 2022-01-29 by dcroote

Raj2032, 2022-01-29:
@dcroote In my opinion in the near future there should be a "write once run everywhere" where it works on other miniboards such as the Odroid etc :)

dcroote, 2022-01-29:
Hi @Raj2032 - I'm not familiar with those boards specifically, but for compiling a single source against an arbitrary target I'd highly recommend [cross](https://github.com/cross-rs/cross). I used it recently to cross-compile this crate for the Raspberry Pi Zero W.

Raj2032, 2022-01-29:
@dcroote I see thanks mate

rahul-thakoor, 2022-01-31:
I think @Raj2032 is looking for a generic hardware abstraction layer for other mcus?

perhaps checkout out https://github.com/rust-embedded/awesome-embedded-rust