rahul-thakoor/rust_gpiozero #38: Some questions

Raj2032, 2022-07-31:
Hi I wanted to ask some questions:

> This library is based on [GPIOZero](https://gpiozero.readthedocs.io/en/stable/index.html) library.

This is from the README.md. So like is this crate using `gpiozero` as the backend or is `rust_gpiozero` a complete rewrite and it uses its own thing?

Does this crate support all the devices as `gpiozero` supports?

Also is this project still continuing?

dcroote, 2022-07-31:
> This is from the README.md. So like is this crate using `gpiozero` as the backend or is `rust_gpiozero` a complete rewrite and it uses its own thing?

This rust crate is inspired by the python GPIOZero package. It's only [dependency](https://github.com/rahul-thakoor/rust_gpiozero/blob/b7d4865dfdb7afdca60a65f4c949f8f5386c624d/Cargo.toml#L17) is rppal for peripheral access.

> Does this crate support all the devices as `gpiozero` supports?

A number of common devices are currently supported, but not everything. Have a look at the [examples directory](https://github.com/rahul-thakoor/rust_gpiozero/tree/master/examples) for some ideas.

> Also is this project still continuing?

Yes.

dcroote, 2022-08-01:
> is this using embedded-hal by any chance?

Not currently, though support for it is a planned feature: https://github.com/rahul-thakoor/rust_gpiozero#features

Raj2032, 2022-08-02:
@dcroote By bringing in embedded-hal, what benefit does it bring to gpio-zero?

dcroote, 2022-08-02:
The Embedded Rust Book [chapter on Portability](https://docs.rust-embedded.org/book/portability/index.html#portability) provides a nice overview and arguments for it.

Raj2032, 2022-08-02:
@dcroote This wouldn't affect the api/syntax of rust_gpiozero?

dcroote, 2022-08-06:
Possibly, but without having spent more time assessing how it could be supported I cannot say either way.

Raj2032, 2022-08-06:
@dcroote Ah right mate.

But if embedded-hal was supported, then would it also be able to support Arduino as mentioned over [here](https://github.com/rahul-thakoor/rust_gpiozero/issues/39)